epics-modules/mrfioc2 #6: cannot compile evg-seq-usage.lyx to a pdf file

jeonghanlee, 2016-04-06:
Hi Michael,

  I've got the following error while compiling evg-seq-usage.lyx file in Debian 8. Do you have any idea about this? 

> 
> Segmentation fault
> /usr/share/lyx/scripts/convertDefault.py ERROR
> Execution of "convert" failed.
> Segmentation fault
> /usr/share/lyx/scripts/convertDefault.py ERROR
> Execution of "convert" failed.
> support/Systemcall.cpp (292): Systemcall: 'python -tt "/usr/share/lyx/scripts/convertDefault.py" "dia:/tmp/lyx_tmpdir.nWOVQSY21366/lyx_tmpbuf0/0_home_jhlee_programs_mrfioc2_documentation_evg-seq.dia" "pdf:/tmp/lyx_tmpdir.nWOVQSY21366/lyx_tmpbuf0/0_home_jhlee_programs_mrfioc2_documentation_evg-seq.pdf"' finished with exit code 1
> Error: Cannot convert file
> ----------------------------------------
> No information for converting dia format files to pdf6.
> Define a converter in the preferences.
> Segmentation fault
> /usr/share/lyx/scripts/convertDefault.py ERROR
> Execution of "convert" failed.
> support/Systemcall.cpp (292): Systemcall: 'python -tt "/usr/share/lyx/scripts/convertDefault.py" "dia:/tmp/lyx_tmpdir.nWOVQSY21366/lyx_tmpbuf0/1_home_jhlee_programs_mrfioc2_documentation_evg-actions.dia" "pdf:/tmp/lyx_tmpdir.nWOVQSY21366/lyx_tmpbuf0/1_home_jhlee_programs_mrfioc2_documentation_evg-actions.pdf"' finished with exit code 1
>

mdavidsaver, 2016-04-21:
Do you have the "dia" package installed?  In lyx, have you tried menu Tools -> Reconfigure?

The following succeeds for me w/ deb8 amd64, lyx 2.1.2-2, dia 0.97.3-1, imagemagick 8:6.8.9.9-5, and probably a whole lot of other bits and pieces that lyx is using without my being aware.

> $ lyx -batch -e pdf evg-seq-usage.lyx 
> 0_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-seq.dia --> 0_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-seq.eps
> 1_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-actions.dia --> 1_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-actions.eps
> 2_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-states.dia --> 2_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-states.eps
> This is pdfTeX, Version 3.14159265-2.6-1.40.15 (TeX Live 2015/dev/Debian) (preloaded format=latex)
>  restricted \write18 enabled.
> entering extended mode
> (./evg-seq-usage.tex
> LaTeX2e <2014/05/01>
> Babel <3.9l> and hyphenation patterns for 2 languages loaded.
> 
> This is pdfTeX, Version 3.14159265-2.6-1.40.15 (TeX Live 2015/dev/Debian) (preloaded format=latex)
>  restricted \write18 enabled.
> entering extended mode
> (./evg-seq-usage.tex
> LaTeX2e <2014/05/01>
> Babel <3.9l> and hyphenation patterns for 2 languages loaded.
> 
> This is dvips(k) 5.994 Copyright 2014 Radical Eye Software (www.radicaleye.com)
> ' TeX output 2016.04.21:1854' -> evg-seq-usage.ps
> </home/mdavidsaver/.texmf-var/fonts/pk/ljfour/jknappen/ec/ecrm1000.600pk>
> </home/mdavidsaver/.texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx1440.600pk>
> </home/mdavidsaver/.texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx1200.600pk>
> </home/mdavidsaver/.texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx1000.600pk>
> </home/mdavidsaver/.texmf-var/fonts/pk/ljfour/jknappen/ec/ecti1000.600pk>
> </usr/share/texlive/texmf-dist/dvips/base/tex.pro>
> </usr/share/texlive/texmf-dist/dvips/base/texps.pro>
> </usr/share/texlive/texmf-dist/dvips/base/special.pro>
> </usr/share/texlive/texmf-dist/dvips/base/color.pro>. 
> </usr/share/texlive/texmf-dist/fonts/type1/public/amsfonts/cm/cmsy10.pfb>[1
> <./0_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-seq.eps>] [2] [3
> <./1_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-actions.eps>] [4
> <./2_home_mdavidsaver_work_bnl_mrfioc2_documentation_evg-states.eps>] [5] [6]

jeonghanlee, 2016-04-22:
Sorry for my stupid question, Michael. Yes, after installing "dia" , I have no issue. Thanks!

closed 2016-04-22 by jeonghanlee