epics-modules/mrfioc2 #17: time2ntp crashes IOC with zero arguments

daykin, 2019-01-29:
A crash occurs if `time2ntp` is invoked in IOC shell with zero arguments. I am using mrfioc2 version 2.1.0.
```
epics> time2ntp
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
```

mdavidsaver, 2019-01-30:
May be fixed by https://github.com/epics-modules/mrfioc2/commit/1e703ba49944cd2137cd04641b80195347d1a404 .

mark0n, 2019-02-19:
v2.2.0 is also affected. Need to test with the change @mdavidsaver mentioned.

mark0n, 2019-02-21:
I can confirm that @mdavidsaver's fix resolves this problem. The exception is now being handled:
```
epics> time2ntp
Error: basic_string::_M_construct null not valid
```
Thanks Michael!

closed 2019-05-09 by mdavidsaver