epics-modules/mrfioc2 #29: Out of date documentation

krmpotic, 2020-02-26:
evr-usage.lyx, chapter 9.10 Database Events is out of date with evrApp/Db/evrevent.db

mdavidsaver, 2020-05-25:
Does anyone find the `EVR Device Support Reference` section a useful reference?  I'm inclined to remove it and shift the useful information someplace closer to the implementation.

mdavidsaver, 2020-06-29:
I think I'll move the device support documentation into the .db files as comments.

krmpotic, 2020-08-20:
I notice flashread, flashwrite usage only in evr-usage.lyx document. There is no mention of using this for upgrading EVM firmware.

Is it possible to flash EVM with these commands? If so, I think we should mention it in the documentation.

jerzyjamroz, 2023-12-15:
The documentation update is ongoing in: https://github.com/epics-modules/mrfioc2/issues/79

closed 2023-12-15 by jerzyjamroz