epics-modules/mrfioc2 #36: Stop installing headers?

mdavidsaver, 2020-05-25:
I'm again considering dropping some or all of the public headers in order to allow more extensive internal restructuring.  Of course this would cause problems for any external users of the API.  So I'd like to find out how much of a problem this would be.

Who is maintaining external code which uses the mrfioc2 APIs?

Which pieces are being used, and to what end?

Links to publicly accessible repositories are much preferred.

dirk-zimoch, 2020-05-25:
mrmDataBufferUser.h and mrmDataBufferType used here:
https://github.com/paulscherrerinstitute/mrfioc2_regDev/blob/007b1739f27f90ddc4053656767c574b01df55a5/src/mrfioc2_regDev.cpp#L89

mark0n, 2020-05-27:
As far as I can tell we aren't using any of these headers at FRIB.