epics-modules/mrfioc2 #51: missing Pause/Abort functionality

krmpotic, 2021-03-08:
evg-usage.lyx mentions Pause(bo) and Abort(bo) records, which are not implemented.

I added a PR https://github.com/epics-modules/mrfioc2/pull/50. Using Disable-Cmd as a template.
Quick test shows that it works OK.

mdavidsaver, 2021-03-08:
pause/abort existed prior to 2.2.0 (circa 2017).  I dropped the feature when reworking the sequencer code to support the sequencers in newer EVRs.  Accounting for the various possible race conditions is complicated, and the alternative is (at best) missing end of sequence notifications.  I wasn't aware of any usage.  Can you elaborate?

krmpotic, 2021-03-22:
@mdavidsaver can you please elaborate regarding your concerns about race conditions? I want to allow user's to abort long-running 10sec+ sequences on EVM.

mdavidsaver, 2021-04-08:
> I want to be able to stop long sequences (e.g. 30 sec+)

I see.  I guess I'm lucking to have avoided this situation so far.

One way I've done this is by splitting a long sequence into pieces, and using the 2 sequencers as a double buffer.  One piece is loading while the previous one is running.  Buffer handling is driven by the start/end of sequence interrupts.  A missing notification will stall this process.  So I can't tolerate missed notifications during "normal" operation.  This approach also has the benefit of allowing a controlled "abort" at the boundary between pieces.

fyi. the double buffer strategy I describe is to my knowledge still in use at FRIB to control pulse repetition rate through a ~10 minute long ramp up process, which may be interrupted into a controlled ramp down.

As I think about it though, an explicit hard abort isn't "normal".  If you are going to allow this, then your application must already be prepared to deal with the unpredictable state which follows.

> I saw that missing end of sequence notifications ... are they something to be relied upon anyway?

As it stand now, yes.