epics-modules/mrfioc2 #57: evgSequencer is gone - what do do about it?

mirek23, 2022-04-30:
I have noticed that the last version of the mrfioc module which contained evgSequencer directory is 2.1.0
Why it was removed? Is it possible to get it back in the current versions?

Best Regards
Mirek

mdavidsaver, 2022-04-30:
Some refactoring was done (in 2017) to support newer EVR cards which include limited EVG functionality (including a sequencer).  The line between "EVR" and "EVG" gets blurry.

https://github.com/epics-modules/mrfioc2/blob/a8cb48d549b7aae13206cf2d168c6f1cfec2e1c4/mrmShared/src/mrmSeq.h#L30-L35

mirek23, 2022-04-30:
Would it work (and would it make sense) if I copied the entire evgSequencer directory from version 2.1.0 to the current one.
The reason I ask is to continue using the code which depends on the functions in evgSequencer.

closed 2023-12-29 by jerzyjamroz