epics-modules/mrfioc2 #66: databuffer protocol ID check

Insomnia1437, 2023-03-17:
When I am using EVR to receive data buffer, I find that the data buffer receiver record is never processed.

After some debugging I found that the problem comes from below code which is inside `evrMrmApp/src/devMrmBufRx.cpp`
```c
  //check protocol id
    if (paddr->proto != 0xff00 && paddr->proto != buf[0]) return;
    if (paddr->proto16 && paddr->proto16 != ntohs(((epicsUInt16*)buf)[0])) return;
    if (paddr->proto32 && paddr->proto32 != ntohl(((epicsUInt32*)buf)[0])) return;
```
My EVG sets the protocol ID to `1`,  but `buf[0]` value is `0`. So the first if-condition is wrongly matched.  When I use `buf[3]` rather than `buf[0]`,  it works fine. Is this a bug or am I just missing something?

The data buffer memory contents from my EVR:
```
epics> pciread 8 0x800 50

0x00000000 0100ff00 ff035700 00000000 0000e600
0x00000010 00000000 c4040e8e 8c00c000 b500beea
0x00000020 80041db2 68028387 40320003 48054e28

epics> pciread 32 0x800 50

0x00000000 00ff0001 000406ff 00000000 00e60000
0x00000010 00000000 8e0e04c4 00b6008c ea1e00bf
0x00000020 b21d0480 87830268 1c003240 214e1e48
```

jerzyjamroz, 2024-02-20:
@Insomnia1437 , is this issue still valid? 
I use the defaults and it works fine:
```
file "mrmevrbufrx.db"
{pattern
{P, OBJ, PROTO, s}
{"\$(P)", "$(EVR):BUFRX", "0xff00", "-"}
}
```

Insomnia1437, 2024-02-20:
Hi @jerzyjamroz ,
Later I realized that this problem is caused by version of mrfioc2. The EVG uses `mrfioc2-2.0.3` (quite old) while my EVR uses `mrfioc2-2.3.0`. I think there exists some compatibility issues. 
We can close this issue.

closed 2024-02-20 by jerzyjamroz