epics-modules/mrfioc2 #69: EVM EVRU and EVRD databuftx.db databuftxCtrl.db and mrmevrbufrx.db

jerzyjamroz, 2023-08-09:
The file: "mrfioc2/evgMrmApp/Db/mtca-evm-300.substitutions" does not set the files: "databuftx.db, databuftxCtrl.db and mrmevrbufrx.db" for EVRU and EVRD. Once it is done:
```
file "databuftx.db"
{pattern
{P, OBJ, PROTO}
{"$(SYS){$(D)U}", "$(EVG):EVRU:BUFTX", 1}
}
file "databuftxCtrl.db"
{pattern
{P, OBJ}
{"$(SYS){$(D)U}", "$(EVG):EVRU:BUFTX"}
}
file "mrmevrbufrx.db"
{pattern
{P, OBJ, PROTO}
{"$(SYS){$(D)U}", "$(EVG):EVRU:BUFRX", "0xff00"}
}
```
The following errors appear:
```
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-send-s8
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-send-u32
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-recv-s8
```
Are "$(EVG):EVRU:BUFTX and $(EVG):EVRU:BUFRX" supported?

mdavidsaver, 2023-09-20:
> Are "$(EVG):EVRU:BUFTX and $(EVG):EVRU:BUFRX" supported?

As I read the current code, I think the objects should be created.  Of course, whether they were or are implemented in firmware is a different question.

https://github.com/epics-modules/mrfioc2/blob/b05a7931f51ec0676931d7f209e91d61bf9c21e0/evgMrmApp/src/evgMrm.cpp#L167-L168

https://github.com/epics-modules/mrfioc2/blob/b05a7931f51ec0676931d7f209e91d61bf9c21e0/evrMrmApp/src/drvem.cpp#L131-L132

fyi. the iocsh functions `dol` and `dor` may be helpful to see if the object name is somehow not what is expected.

jerzyjamroz, 2023-09-21:
- The objects are there (see below).
- Following the message: "_Missing required option OBJ_", I have tried to change PROTO as well but without any result (I think, there are no other options).
- The same errors persist.

```
# dol
Object: EVG:EVRU
Type: 6EVRMRM
Object: EVG:EVRU:BUFRX
Type: 8mrmBufRx
Object: EVG:EVRU:BUFTX
Type: 12mrmDataBufTx
Object: EVG:EVRU:FPIn0
Type: 8MRMInput

Object: EVG:EVRD
Type: 6EVRMRM
Object: EVG:EVRD:BUFRX
Type: 8mrmBufRx
Object: EVG:EVRD:BUFTX
Type: 12mrmDataBufTx
Object: EVG:EVRD:FPIn0
Type: 8MRMInput
```
```
# dor
Object: EVG:EVRU:BUFRX
Type: 8mrmBufRx
  b Enable = 1
Object: EVG:EVRU:BUFTX
Type: 12mrmDataBufTx
  b Enable = 1
  j Max length = 2044
  b Ready to send = 1

Object: EVG:EVRD:BUFRX
Type: 8mrmBufRx
  b Enable = 1
Object: EVG:EVRD:BUFTX
Type: 12mrmDataBufTx
  b Enable = 1
  j Max length = 2044
  b Ready to send = 1

```
```
# iocsh errors
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-send-s8
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-send-u32
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-recv-s8
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:U-dbus-recv-u32
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:D-dbus-send-s8
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:D-dbus-send-u32
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:D-dbus-recv-s8
Missing required option OBJ
recGblRecordError: Couldn't parse link string No device at specified address PV: TDL-JJ:Ctrl-EVG-1:D-dbus-recv-u32
```