epics-modules/mrfioc2 #79: Move documentation to github pages

gabrielfedel, 2023-09-22:
Right now the mrfioc2 documentation is located on sourceforge, where only few people have permission to do updates. 
A good solution is move the documentation to github pages.

gabrielfedel, 2023-09-22:
I couldn't yet make the automatic generation (some issues with the lyx files and the new doxygen), so I suggest we push it manually for now, this way we can retire the sourceforge.

It is possible to see how the documentation would be here: https://gabrielfedel.github.io/mrfioc2/

I have pushed the same version to the branch gh-pages and if it is fine we just need to set it as the source of the github pages.

@mdavidsaver and @jerzyjamroz  could you please take a look here?

jerzyjamroz, 2024-01-15:
It would be good to protect ```gh-pages``` branch against deletion.

gabrielfedel, 2024-01-17:
The documentation is on github pages now on: https://epics-modules.github.io/mrfioc2/
The generation is not automatic yet, so updates need to be pushed to the right branch (gh-pages)

closed 2024-01-17 by gabrielfedel