epics-modules/mrfioc2 #89: Add support for MTCA EVR 300U and 300RF

zioven, 2023-10-09:
Hi,

Some time ago my coworker provided pull request #63   for the 300U and 300RF versions of the EVR.

Can somebody provide guidance what is needed to include those two versions into main release?

Best regards,

Žiga

jerzyjamroz, 2023-10-10:
Hi, I tried to upload first the naming agreements as in the future, the mrfioc2 should be able to follow unified naming standards. One of the modifications looks like this: https://github.com/epics-modules/mrfioc2/pull/63/commits/2d49873cae13510d46edbeae3373c4a989cae759
It means: $(SYS){$(D)}Xxx:Yyy:1 is changed now to: $(P)XxxYyy1. All the previous .substitutions files are still valid (the backward compatibility is kept).

jerzyjamroz, 2023-10-10:
I tried a preliminary run of your implementation merged to a test master and I got ```6331 Segmentation fault``` while just master works fine. I will investigate if it is my setup or what is the reason. I think It would be good to rebase this PR to the actual master as well.