epics-modules/mrfioc2 #96: To align handling of the GDX outputs.

jerzyjamroz, 2023-10-13:
* https://github.com/epics-modules/mrfioc2/pull/87
* https://github.com/epics-modules/mrfioc2/pull/95
* https://github.com/epics-modules/mrfioc2/pull/94
VS
* https://github.com/epics-modules/mrfioc2/pull/63

zioven, 2023-10-13:
missing reference to commit: 64b576325f4c8771b07360b42ee5d721a6a326ac
This commit provides the information so the driver maps the Inputs 16 and 17 as TCLKA and TCLKB (from MRF MTCA manual)
[image]

jerzyjamroz, 2023-10-16:
@zioven , I did extensive CLKA tests today and I think my impelemtation is the way to go. It works as expected.

In summary:
* I will wait for you comment, once ok, I will merge this issue PRs (my part).
* Then, let's aggree the names for the EVR models within "mrmEvrSetupPCI" command.
* Then let's merge your code.

If anything, all ideas are welcome.

zioven, 2023-10-17:
@jerzyjamroz Can you elaborate on the tests performed and what shortcomings had the second solution?

I am just curious and would like to understand if and how the performance is affected by both solutions.

jerzyjamroz, 2023-10-17:
@zioven , the patches included here were used to run the CLKA output with 40-bit granularity (1 to 40-bit test was done successfully). The performance was on the level of ps. If you want more info, let's have a chat.

jerzyjamroz, 2023-10-26:
@zioven , I will merge PR #94 and #95 at the end of the day. If you are too busy then I will merge the @ZanMaticPratnemer code with those PRs included in it.

closed 2023-12-27 by jerzyjamroz