epics-modules/mrfioc2 #121: Can't use upstream event between EVRU and EVM

agaget, 2024-01-19:
It was working before, but now I didn't manage to make this work.
[image]

It works on this commit, it's from the old fork icshwi  :
https://github.com/icshwi/mrfioc2/commit/15d003d2899297eecebf205621f697a43bb446be
We were using a fork of this commit (adapted to our old environment)

The EVM and EVM-EVRU works fine, I can send events through EVG, I can simulate front edge detect by Univ8 with "Force High/Low". 
[image]

The EVR Input works, because I can detect the event 213 on local
[image]

It means that the problem comes from EVR : the Upstream event that should send 13  upstream, or the pulse of the EVRU that should react to 13 but didn't.

I hope it's clear ...

I'm still digging

agaget, 2024-01-19:
if it can help I apply the following configuration (with the naming back then)
```
caput $(P_EVM):Mxc0-Prescaler-SP 88000000
caput $(P_EVM)U:DlyGen0-Width-SP     100
caput $(P_EVM)U:DlyGen0-Evt-Trig0-SP  13
caput $(P_EVM)U:OutFP0-Src-Pulse-SP "Pulser 0"
caput $(P_EVM):TrigEvt0-TrigSrc-Sel  Mxc0
caput $(P_EVM):TrigEvt0-EvtCode-SP 1
caput $(P_EVM):TrigEvt1-TrigSrc-Cont-Sel Univ8
caput $(P_EVM):TrigEvt1-EvtCode-SP  13

caput $(P_EVR):OutFP0-Src-Pulse-SP 0
caput $(P_EVR):DlyGen0-Width-SP 500000
caput $(P_EVR):DlyGen0-Evt-Trig0-SP 1
caput $(P_EVR):In0-Trig-Ext-Sel "Edge"
caput $(P_EVR):In0-Trig-Back-Sel "Edge"
caput $(P_EVR):In0-Code-Back-SP 13
caput $(P_EVR):In0-Code-Ext-SP 213
caput $(P_EVR):EvtA-SP.OUT "@OBJ=EVR-MTCA,Code=1"
caput $(P_EVR):EvtB-SP.OUT "@OBJ=EVR-MTCA,Code=2"
caput $(P_EVR):EvtC-SP.OUT "@OBJ=EVR-MTCA,Code=213"
caput $(P_EVR):EvtD-SP.OUT "@OBJ=EVR-MTCA,Code=13"

caput $(P_EVR):In0-Trig-Back-Sel  Edge
caput $(P_EVR):In0-Code-Back-SP  13
caput $(P_EVR):In0-Trig-Ext-Sel  Edge
caput $(P_EVR):In0-Code-Ext-SP  213
```` 
If it works we can see directly counter A,C,D increasing (corresponding to event 13), if not only A,C

Here I have plugged on the EVR a LEMO wire between FP0 and IN0.
It work the same way withou wire if we use Backplane Input an outputs (need to add it in the subs file for now)

jerzyjamroz, 2024-01-22:
@agaget , I guess the issue is related to:
```
"TrigEvt1EvtCode-SP" "13"
"TrigEvt1TrigSrc1-Sel" "Univ8"
```

agaget, 2024-01-22:
> @agaget , I guess the issue is related to:
> 
> ```
> "TrigEvt1EvtCode-SP" "13"
> "TrigEvt1TrigSrc1-Sel" "Univ8"
> ```
No I don't think, in the embedded EVRU if I "force low" "force high", the Univ 8. It triggers well the event 13.

[image]

I think it's more a ***bug*** than a ***question***. The same configuration works with a lower version of the driver.

zioven, 2024-01-22:
@agaget I haven't used this with our setups, but do you need to enable event forwarding on the EVM embedded EVRs?

https://github.com/epics-modules/mrfioc2/blob/master/evrMrmApp/src/drvemSetup.cpp#L971

jerzyjamroz, 2024-01-22:
@agaget , remove the intermediate EVM and check if all the PVs are properly applied.

agaget, 2024-01-22:
> No I don't think, in the embedded EVRU if I "force low" "force high", the Univ 8. It triggers well the event 13. [...]

Sorry the synoptic I've made was jsut a copy/paste from one of my powerpoints, the real case is this one, I don't have intermediate EVM :
[image]

>@agaget I haven't used this with our setups, but do you need to enable event forwarding on the EVM embedded EVRs?
>
>https://github.com/epics-modules/mrfioc2/blob/master/evrMrmApp/src/drvemSetup.cpp#L971
I'm not sure what you asking, but I will look into it

agaget, 2024-01-22:
> @agaget I haven't used this with our setups, but do you need to enable event forwarding on the EVM embedded EVRs?
> 
> https://github.com/epics-modules/mrfioc2/blob/master/evrMrmApp/src/drvemSetup.cpp#L971

I have this 
```
EVR SBCT > mrmEvrForward("EVR-MTCA5")
Events forwarded: 
EVR SBCT >
```
But with the old version of the driver it works even if no events are marked as forwarded.
I have tried anyway 
```
EVR SBCT > mrmEvrForward("EVR-MTCA","all")
EVR SBCT > mrmEvrForward("EVR-MTCA")      
Events forwarded: 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 
```
but still don't work.

Note that if you want to test it quickly, with my pull request, you can use BPIn0 instead of In0 for the EVR. It should work the same without the wire

agaget, 2024-01-24:
Ok I've found where is the problem, I'm not sure why it is occuring. In the pulser of the embedded EVR the setWidth function doesn't work correctly and set the width to 0 whatever what we put.

The clock of the embedded EVR is always 0, and so the width can't be computed correctly, if I comment the line and put my frequence manually, it works fine.

https://github.com/epics-modules/mrfioc2/blob/9abc5163e86251db3d08f814dad45dabfd1baf41/evrMrmApp/src/drvemPulser.cpp#L117

```
void
MRMPulser::setWidth(double v)
{
    double scal=double(prescaler());
     
    double clk=88000000;

    //double clk = (owner.clock() == 0.0) ? owner.clockTS() : owner.clock(); // in MHz. MTicks/second
    if (scal<=0) scal=1;

    epicsUInt32 ticks=roundToUInt((v*clk)/scal);

    printf("test setWidth %f %f %f \n",scal, clk,ticks);
    setWidthRaw(ticks);
}
```
The line has been modified by @jerzyjamroz 3 month ago, but the old line doesn't work also anyway. I think the problem comes from elsewhere. The clk var indicates 0 even if I force the frequency through the PV Labo:titan-EVMU:LinkClk-SP

jerzyjamroz, 2024-01-25:
@agaget , this line should prevent that scenario but it seems it is more complex. I will debug it soon.

jerzyjamroz, 2024-01-29:
@agaget , you forgot to set: ```U-TimeClock-SP 88.0525```.

agaget, 2024-01-29:
hmmm, ok. I've never defined it anywhere. It was working fine.
In my old driver, I have a macro $(FEVT) in the VAL field of the TimeClock-SP that explain why I never had to set this variable.

But with the "modern" driver, in my MTCA EVR300U I don't defined it through this variable and it's ok anyway.

In my MTCA EVR 300U, I have this
```
EVR Labo Titan> dbpr Labo:titan-EVR:TimeClock-I
AMSG:               ASG :               DESC: Timestamp tick rate               
DISA: 0             DISV: 1             NAME: Labo:titan-EVR:TimeClock-I        
NAMSG:              RVAL: 0             SEVR: NO_ALARM      STAT: NO_ALARM      
SVAL: 0             TPRO: 0             VAL : 88            
EVR Labo Titan> dbpr Labo:titan-EVR:TimeClock-SP
AMSG:               ASG :               DESC: Timestamp tick rate               
DISA: 0             DISV: 1             NAME: Labo:titan-EVR:TimeClock-SP       
NAMSG:              OMOD: 0             OVAL: 0             RBV : 0             
RVAL: 0             SEVR: NO_ALARM      STAT: NO_ALARM      TPRO: 0             
VAL : 0             
EVR Labo Titan> 

```
I don't know how it's possible. 88 is defined elsewhere for the link.

I think there is a lack of feedback concerning this issue,  the next one with this issue will have the same clue to solve it.

jerzyjamroz, 2024-01-30:
@agaget , I will add your feedback to the .db files.

closed 2024-01-30 by jerzyjamroz