epics-modules/mrfioc2 #147: Pulse generation problem on version 2.6.0 onwards on EVR evr-mtca-300u.uv?

amichelotti, 2024-04-14:
Hello,
I started having problem on pulse generation since my last update. I was on version 2.3.0., I tried to align to latest version but no way to output on FP0,0.. pulser0,1... on my EVR mtca 300U. I use in th evr-mtca-300u.uv.db in the ioc configuration and as OPI I use the evr/evr-mtca-300u.bob.  The events are correctly received (counters increments) but sinze version 2.6.0 I cannot see the pulser on the FP output. The same configuration works till version with tag 2.5.0. I saw lots of changes between 2.5.0 and 2.6.0, maybe I missed something ?

jerzyjamroz, 2024-04-15:
Did you try this property: ```OutFP0SrcPulse-SP``` ?

amichelotti, 2024-04-18:
yes of course. I set to Pulser0. And the pulser is correctly set with a width and delay, the corresponding event counter (in the expert panel) is correctly incrementing. I did the same procedure with version <=2.5.0 (works), 2.6.x (no)

Insomnia1437, 2024-04-18:
Hi Andrea! 

Could you check the value of these PVs? 
```
$(P)OutFP0Ena-SP
$(P)OutFP0Src-RB
$(P)DlyGen0Ena-Sel
$(P)DlyGen0Delay-SP
$(P)DlyGen0Width-SP
$(P)DlyGen0EvtTrig0-SP
$(P)DlyGen0EvtTrig1-SP
$(P)DlyGen0EvtTrig2-SP
```
Also, are there any error messages from the iocsh?

amichelotti, 2024-04-18:
I found something weird if you do:
grep SrcGate evrMrmApp/opi/*
grep SrcGate opi/bob/evr/* <<<- here a match I'm using that one

Which opi should I use for EVR?

Insomnia1437, 2024-04-18:
The latter one is generated from this PR #132 recently
My recommendation is to use that one, but it also contains a few minor problem, you may also need to change some PV names to match the db file.

amichelotti, 2024-04-19:
Hi all, now I reverted to master to show what happens with snapshots. I'm using the bob/evr/evr-mtca-300u.bob.
I attach snapshot of the trials with values. With this version seems not working Pulser output. Force high/low and prescaler work fine. iocsh shows this warning  :

TS out of bounds 662230f0 3b9ba936 oflow 57142 evrMrmTimeNSOverflowThreshold 0 diff 57142
TS out of bounds 662230f5 3b9acc98 oflow 664 evrMrmTimeNSOverflowThreshold 0 diff 664
TS out of bounds 662230f5 3b9acc98 oflow 664 evrMrmTimeNSOverflowThreshold 0 diff 664
TS out of bounds 662230fb 3b9c521b oflow 100379 evrMrmTimeNSOverflowThreshold 0 diff 100379
TS out of bounds 662230fb 3b9c521b oflow 100379 evrMrmTimeNSOverflowThreshold 0 diff 100379

[image]
[image]
[image]
[image]
[image]

amichelotti, 2024-04-19:
Hi !! thank you very much for your support! The version seems more or less aligned (you have 207.e I have 207.11) and also the setup. However respect your clean green panels  I've something in alarm (I attach the screenshot). And the time is flipping between synchronized and not synchronized (that produces also logs in the ioc). I was thinking the problem is something more tricky. In my setup Generator and receivever are on the same utca controller  (1 generator and 2 evr) and are connected through a fiber of 5m long. Since the difference are in Delay compensation and on the logs on the ioc I wouldn't want it to be 'time problem' normally hard to debug. And it could also explain why in the 2.5.0 apparently I don't have this issue on the pulser output. What do you think could be the cause of this difference (and also these alarms)?
[image]

amichelotti, 2024-04-20:
Thanks, I followed your instructions but the periodic timing disconnections are still there. Where can I see the changes between firmware 207.e and 207.11? However I would like to know if I didn't miss something in the ioc/driver/configuration of the releases 1.6.x:
ioc:
#!../../bin/linux-x86_64/mrf

## Register all support components
dbLoadDatabase("../../dbd/mrf.dbd")
mrf_registerRecordDeviceDriver(pdbbase)
mrmEvrSetupPCI("EVR0","06:00.0","UNIV")
mrmEvrSetupPCI("EVR1","0e:00.0","UNIV")
mrmEvgSetupPCI("EVG","0c:00.0")
dbLoadRecords("../../db/evr-mtca-300u.uv.db","P=__IOC_PREFIX__:evr:, EVR=EVR0")
dbLoadRecords("../../db/evr-mtca-300u.uv.db","P=__IOC_PREFIX__:evr1:, EVR=EVR1")
dbLoadRecords("../../db/evm-mtca-300.uv.db","P=__IOC_PREFIX__:evg:, EVG=EVG")
iocInit()
----------
Kernel module:
from mrmShared
lsmod:
mrf                    18421  3 
uio                    19338  7 mrf
parport                46395  3 mrf,ppdev,parport_pc
-----

Thanks a lot again.

Insomnia1437, 2024-04-20:
Here is FW version change log. If you want to know the details of FW change, you need to contact the MRF company 
http://www.mrf.fi/index.php/topcontact

Based on the code you provided, I don't see any obvious issues.
[image]

amichelotti, 2024-04-20:
Hi all, I'm trying to debug deeper and deeper. Now I found something strange looking the start the iocs logs that for my fault where congestioned of prints and I did't catched that it gives a warning for the minimum FW version even if in the OPI we see an acceptable version :
```
mrmEvrSetupPCI("EVR0","06:00.0","UNIV")
Notice: devPCIFindSpec() expect B:D.F in hex
Device EVR0  6:0.0 slot=3
Using IRQ 151
Config for EVR FP UNIV model (mTCA-EVR-300U).
FWVersion 0x18110207
Found version 519
Found SFP EEPROM
Sequencer capability detected
mTCA:  Model mTCA-EVR-300U Pul 24 PS 8 OFP 4 OFPUNIV 4 ORB 10 OBack 8 OFPDly 2 CML 2 Kind 0 IFP 32
EVR FIFO task start
Enabling interrupts
mrmEvrSetupPCI("EVR1","0e:00.0","UNIV")
Device EVR1  e:0.0 slot=1
Using IRQ 153
Config for EVR FP UNIV model (mTCA-EVR-300U).
FWVersion 0x18140207
Found version 519
Found SFP EEPROM
Sequencer capability detected
mTCA:  Model mTCA-EVR-300U Pul 24 PS 8 OFP 4 OFPUNIV 4 ORB 10 OBack 8 OFPDly 2 CML 2 Kind 0 IFP 32
EVR FIFO task start
Enabling interrupts
mrmEvgSetupPCI("EVG","0c:00.0")
Device EVG  c:0.0
Using IRQ 152
FPGA version: 280e0207
mTCA-EVM-300 #Inputs FP:3 UV:16 RB:0
EVM automatically creating 'EVG:FCT', 'EVG:EVRD', and 'EVG:EVRU'
Found SFP EEPROM
Found SFP EEPROM
Found SFP EEPROM
Found SFP EEPROM
Found SFP EEPROM
Found SFP EEPROM
Found SFP EEPROM
Found SFP EEPROM
Warning: Recommended minimum firmware 2 version is 207.6, found 207.0
Found SFP Strangeness 00000000
Sequencer capability detected
VME 64:  Model mTCA-EVM-300 (EVRD) Pul 16 PS 3 OFP 8 OFPUNIV 0 ORB 0 OBack 0 OFPDly 0 CML 0 Kind 1 IFP 8
EVR FIFO task start
Warning: Recommended minimum firmware 2 version is 207.6, found 207.0
Found SFP Strangeness 00000000
Sequencer capability detected
VME 64:  Model mTCA-EVM-300 (EVRU) Pul 16 PS 3 OFP 8 OFPUNIV 0 ORB 0 OBack 0 OFPDly 0 CML 0 Kind 1 IFP 8
EVR FIFO task start
PCI interrupt connected!
```
Can be the problem? in this case can you point me on how to update the FW?

Insomnia1437, 2024-04-20:
This warning is generated from the embedded EVRs (for mrfioc2, their names are EVRU and EVRD) inside EVM, not the normal EVR. The OPI shows the FW version of the normal EVR.

You can read this if you are interested in the embedded EVRs 
https://indico.cern.ch/event/766611/sessions/305379/attachments/1855185/3046845/MRF.TimingWS.A.pdf

If you want to update the FW, you can use `flashwrite` command, the function is defined [here](https://github.com/epics-modules/mrfioc2/blob/master/mrfCommon/src/flashiocsh.cpp)
e.g.,
`flashwrite("EVR0:FLASH", 0, "mTCA-EVR-300DC.207.6.bit")`

amichelotti, 2024-04-21:
Hi! at last my main issue disappeared. I restarted from a fresh repo (), I updated:
epics-R7.0.7 -> epics-R7.0.8
devlib2 2.11 -> 2.12

I would to thank you very much for your support!!, that convinced me that something was wrong on my side. I'm still curious to understand what went wrong in my previous setup, but I've not time at all.

closed 2024-04-21 by amichelotti