epics-modules/mrfioc2 #148: AC trigger generates Events?

amichelotti, 2024-04-30:
Hello,
now I'm retrying some tests with AC trigger with this new release. 
If I set trigevent0 source AC  with event code 1 and I set Divider, syncsrc (Event Clk) of Ac trigger in my EVG , I would expect the EVR can see  the event code '1'. But I dont see any event '1' arriving. What I miss?
[image]

Insomnia1437, 2024-05-01:
Can you verify your 50 Hz AC TTL level square wave signal using oscilloscope? This signal should be provided to the IN0 input on EVM.

amichelotti, 2024-05-07:
Hi, I checked square and levels, but no way to make it working, until I've done:
```bash
kill <IOCPID>
rmmod mrf
modprobe pci_mrfevg
modprobe pci_mrfevr

./EvgSetACInput /dev/ega3 0 0 1 0
./EvgSetACMap /dev/ega3 0
./EvgSetTriggerEvent /dev/ega3 0 1
rmmod pci_mrfevg
rmmod pci_mrfevr
modprobe mrf
<startioc>
```
I've also noted that linux driver 'pci_mrfevg.ko' cannot be used (doesnt generate /dev/egaxx ) if 'mrf.ko' driver is loaded.
This way I see the events. So it seems that pci_mrfevg.ko (the above utilities ) does something different respect accessing PVs that should select and activate the AC trigger.
I also noted that the led below the IN0 is always OFF independently from AC trigger is working or not.

amichelotti, 2024-05-31:
Thanks last patch resolved the issue

closed 2024-05-31 by amichelotti