epics-modules/mrfioc2 #172: lutX should not be used

Insomnia1437, 2024-07-10:
When mapping EVR output to `Prescaler 1`, OPI reported `READ_ALARM`. real output is not affected

It looks like the info field of `lutX` should be reserved as an "unknown" option. 

I am looking into it and will fix it when I have time

https://github.com/epics-modules/mrfioc2/blob/5b0c1a1739a210ffc8bfe1b1118723cdd9fd6ee2/mrfCommon/src/devlutstring.cpp#L90-L96

https://github.com/epics-modules/mrfioc2/blob/5b0c1a1739a210ffc8bfe1b1118723cdd9fd6ee2/evrMrmApp/Db/mrmevrout.db#L439-L446

See

[image]

jerzyjamroz, 2024-08-22:
@Insomnia1437 , as far I can notice quickly, [mrfioc2/mrfCommon/src/devlutstring.cpp](https://github.com/epics-modules/mrfioc2/blob/5b0c1a1739a210ffc8bfe1b1118723cdd9fd6ee2/mrfCommon/src/devlutstring.cpp#L90-L96) Lines 90 to 96 in [5b0c1a1](https://github.com/epics-modules/mrfioc2/commit/5b0c1a1739a210ffc8bfe1b1118723cdd9fd6ee2) can be deleted. It looks like an artifact, what is the purpose of that?

Insomnia1437, 2024-08-23:
As Michael explained, I think it's more appropriate to keep this customization. I suggest changing the hard-coded `lutX` to `lutXX` to increase future scalability.

jerzyjamroz, 2024-08-23:
Ok, so I propose PR https://github.com/epics-modules/mrfioc2/pull/175 with lutX changed to lut_ to avoid the letters.