epics-modules/mrfioc2 #180: EVM input mask record name and driver function name

hongran, 2024-09-23:
I have one concern about the Hw masks implemented in https://urldefense.us/v3/__https://github.com/epics-modules/mrfioc2/pull/124__;!!G_uCfscf7eWS!broV99Q0qicXStU7HxjH-zQqBKfbM2Lopx3IzjjpDqvdR-r9a_qV0BToBEdrucnsfMxTmZGMpuT8ymzRSSc$

The evg register names EVG_INP_FP_MASK  and EVG_INP_FP_MASK _shift, the record name FPMask-Sel indicates that these are for front-panels only. However, they can be applied to any hardware inputs. Also, they only mask the sequencer events. 

Can we change the FP into SQ in these names to avoid this confusion?

agaget, 2024-09-24:
Hello,
It's been a long time since I did that (merged only few month ago, but implemented 4 or 5 years ago) so I'm not totally into the code, but I've looked in the doc and it's called FPxSQMky

[image]

There is a clear difference with UIxSQMK
[image]

Not clear from me what is UIxSQMK, maybe it depends on the EVM used....
For MTCA I suppose the Univ0 to Uni15 are considered to be FP input for the internal EVG ? why not UI Input ??
[image]

To conclude we can add SQ to add information but not remove FP

hongran, 2024-09-24:
@agaget Hello, thanks for your reply. 
If I understand it correctly, these generic functions can be applied to all types of input, not just the FP. EVG_INP_FP_MASK is a shift relative to a base address of an output channel. For both FP and UI, the MASK is bit 31-28, the difference is the base address. 
As you pointed out, Univ Input for EVM-300 are internally wired for interfacing with the EVRU and EVRD. 

Besides FP and UI, there are also transition board inputs that can be used as the MASK source:
[image]

The transition board is on the rear side, so using FP in it may cause confusions. That's my main concern.

Thanks!