epics-modules/mrfioc2 #200: "TS out of bounds error"

daykin, 2024-12-13:
Hello,
I am getting the following error using an MRF CPCI-EVR-300, getting events from a CPCI-EVG-300, about once every 1.5 minutes:

TS out of bounds 675cba55 3e0a2da2 oflow 40854434 evrMrmTimeNSOverflowThreshold 0 diff 40854434
TS out of bounds 675cba62 3e0a3bb5 oflow 40858037 evrMrmTimeNSOverflowThreshold 0 diff 40858037
TS out of bounds 675cba8e 7327f5a9 oflow 931998633 evrMrmTimeNSOverflowThreshold 0 diff 931998633

System configuration:
80.5MHz carrier, EVG time ticks on an external 1pps signal

mrfioc version 2.6.1, EVR firmware 00b.2

MRF kernel module version 6.1.0-23-rt-amd64

Is there a common cause of this issue?

mdavidsaver, 2024-12-14:
Well, this is strange...

Are other EVRs also giving this error?

Firstly, check to make sure this EVR isn't loosing lock.

Then setup this EVR, and another, with an event counter on code 125 and see that they tick regularly, and in sync.

daykin, 2024-12-16:
Now that I look at it, it's even weirder. Other EVRs are seeing it, so it must be an EVG problem of some kind.  Every time this hiccup happens, the timestamp distributed to devices jumps forward by a few seconds. After running for the weekend, it's about 2.5 hours ahead of reality. But there is no output from the EVG indicating an error. With two EVRs, they both go haywire at the same time. It seems the 125 event is having some crazy arrhythmia. Note the timestamps, the EVR-provided time goes nuts.
```GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:19.999999 1  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:20.999999 2  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:21.999999 3  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:22.999999 4  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:23.999999 5  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:24.999999 6  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:25.999999 7  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:26.999999 8  <<<Starting here
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 9  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 10  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 11  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 12  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 13  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 14  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 15  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 16  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 17  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 18  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:27.559382 19  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:38.999999 20  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:39.999999 21  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:40.999999 22  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:41.999999 23  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:42.999999 24  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:43.999999 25  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:44.999999 26  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:45.999999 27  
GTS_FTS:EVR_N0003:CNT_CALC_EVTA 2024-12-16 12:34:46.999999 28  
```
The same, with system time:
```
test_w_system_time             2024-12-16 10:04:56.353548 1  
test_w_system_time             2024-12-16 10:04:57.353347 2  
test_w_system_time             2024-12-16 10:04:58.356070 3  
test_w_system_time             2024-12-16 10:04:59.353428 4  
test_w_system_time             2024-12-16 10:05:00.353314 5  
test_w_system_time             2024-12-16 10:05:01.353351 6  
test_w_system_time             2024-12-16 10:05:02.353512 7  
test_w_system_time             2024-12-16 10:05:03.353528 8 <<<Starting here 
test_w_system_time             2024-12-16 10:05:03.912758 9  
test_w_system_time             2024-12-16 10:05:05.094541 10  
test_w_system_time             2024-12-16 10:05:06.094424 11  
test_w_system_time             2024-12-16 10:05:07.094458 12  
test_w_system_time             2024-12-16 10:05:07.457911 13  
test_w_system_time             2024-12-16 10:05:08.094411 14  
test_w_system_time             2024-12-16 10:05:09.094494 15  
test_w_system_time             2024-12-16 10:05:10.094571 16  
test_w_system_time             2024-12-16 10:05:11.094463 17  
test_w_system_time             2024-12-16 10:05:12.094448 18  
test_w_system_time             2024-12-16 10:05:13.094575 19  
test_w_system_time             2024-12-16 10:05:14.094447 20  
test_w_system_time             2024-12-16 10:05:15.094477 21  
test_w_system_time             2024-12-16 10:05:16.094493 22  
test_w_system_time             2024-12-16 10:05:17.094449 23  
test_w_system_time             2024-12-16 10:05:18.094441 24  
test_w_system_time             2024-12-16 10:05:19.096664 25  
test_w_system_time             2024-12-16 10:05:20.094493 26  
test_w_system_time             2024-12-16 10:05:21.094468 27  
test_w_system_time             2024-12-16 10:05:22.094494 28  
test_w_system_time             2024-12-16 10:05:23.062527 29  
```

This is reproducible and happens at identical times on multiple EVRs, so I will look at the EVG 1pps and let you know what I find out.

daykin, 2024-12-16:
OK, turns out it's the 1pps source that's going nuts. Nothing wrong with mrfioc2 in this case. Thank you for your assistance!